ilpincy/argos3 #176: Lua and QT version after MacOS brew upgrade

freedomcondor, 2021-05-25:
Hi, Carlo,

        I'm working with argos3 on a Macbook. Today I did "brew upgrade", and it automatically upgraded my lua to lua5.4, and qt to qt6. After that, when cmaking, argos is able to find lua5.4 and qt6. 
        Lua5.4 hasn’t shown its problem yet. 
        However, for qt6, after I recompiled and installed when I try to run argos, it says :
               “Sorry, argos3 cannot run on this computer. Qt requires macOS 11.0.0 or later, you have macOS 10.16.0.”
        That doesn't sound right, because I just upgraded my system to 11.3.1.

        I tried to downgrade qt back by "brew remove qt", and "brew install qt5", but then argos don't find qt when cmaking.
        I made a trick by changing this line from qt to qt@5 :
              https://github.com/ilpincy/argos3/blob/master/src/cmake/FindARGoSQTOpenGL.cmake#L43
        and then everything works. (with qt5 and lua5.4)

        I also tried to downgrade lua to lua5.3, but I cannot manage to make argos find lua@5.3.

        I guess the qt6 problem is between qt and macOS. Before they sort it out, do we have a better way to switch to qt5?
        Also for lua, we haven't test our code with lua5.4, it would be safer to work with lua5.3. Is there a way to switch to lua5.3?

        Best
Weixu

allsey87, 2021-05-25:
Some relevant links:
1. https://forum.qt.io/topic/121255/cannot-build-qt5-on-macos-11-0-1-big-sur-with-macports-as-the-version-information-is-mistaken
2. https://bugreports.qt.io/browse/QTBUG-85546

ilpincy, 2021-05-25:
Thanks for the bugreport @freedomcondor! I don't have Big Sur on my Macs, so it's useful to know that compilation doesn't work.

To fix the problem, once you install Qt5 to circumvent the bugs of Qt6, you don't need to change anything in ARGoS. Just set the variable `ARGOS_BREW_QT_CELLAR`. So, to force `qt@5`, you simply need to run
```
cmake -DARGOS_BREW_QT_CELLAR=$(brew --cellar qt@5) ...
```
The code is written to honor the value of that variable if it's set (see [here](https://github.com/ilpincy/argos3/blob/master/src/cmake/FindARGoSQTOpenGL.cmake#L40)).

ilpincy, 2021-05-25:
Regarding Lua 5.4, I don't see the need to revert to Lua 5.3. [This check](https://github.com/ilpincy/argos3/blob/master/src/cmake/ARGoSBuildChecks.cmake#L116) only asks for Lua >= 5.3, so 5.4 and later are fine.

ilpincy, 2021-05-25:
If the issue is with your own code, just uninstall Lua 5.4 from your system and install Lua 5.3 instead. ARGoS won't mind, as long as you recompile it.

freedomcondor, 2021-05-25:
Thank you for the tips about `-DARGOS_BREW_QT_CELLAR=$(brew --cellar qt@5)`. It works!
For lua5.3, the problem is that on macOS brew considers lua5.3 as an outdated version and name it lua@5.3, and it cannot be found by find_package(lua).

ilpincy, 2021-05-26:
> For lua5.3, the easiest way is to unlink lua and link lua@5.3 (brew unlink lua, brew link lua@5.3), then lua5.3 will be the default lua and found by argos.

Yes, this is an acceptable workaround.

> it looks like https://github.com/ilpincy/argos3/blob/master/src/cmake/FindARGoSQTOpenGL.cmake#L63
should end its IF before https://github.com/ilpincy/argos3/blob/master/src/cmake/FindARGoSQTOpenGL.cmake#L47 so that ARGOS_BREW_QT_CELLAR provided either by user or by execute_process can be effective in either case, right?

You're right indeed! I'll fix it as you suggest.