ilpincy/argos3 #177: argos3_simulator.pc no longer generated ?

jharwell, 2021-06-11:
With the latest master, the pkg-config file for ARGoS does not seem to be generated, so when I try to use the ARGoS cmake module elsewhere, I get all sorts of cryptic errors.

My cmake command is just `cmake ../src` from `<argos>/build`.

Am I doing something wrong ?

ilpincy, 2021-06-13:
The compilation scripts changed a few months ago - check the updated [examples](https://github.com/ilpincy/argos3-examples) to see how to correctly detect ARGoS in your system. Most of the configuration is done automatically now.

jharwell, 2021-06-14:
That was not the issue. The issue was that the argos3_simulator.py file is not _generated_ by the ARGoS build process, not that it is generated and my code can't find it when I try to use the ARGoS cmake package in my code later.

Steps to reproduce:

1. Remove `$HOME/.local/lib/pkg-config` directory if it exists.
1. Checkout latest ARGoS master 7c8a51b63400ccf477d5d5132b83349d81d1ae1f
2. `cmake -DCMAKE_INSTALL_PREFIX=$HOME/.local -DARGOS_INSTALL_LDSOCONF=NO ../src` from `<argos>/build`
3. `make && make doc && make install`

Expected result: `$HOME/.local/lib/pkg-config/argos3_simulator.pc` exists after installation. 
Actual result: File does not exist.

Prior to commit c0ff091e9f037b022281cec6080c78fa3bd57879 this does not happen and everything works as expected. 

It looks like the following lines were deleted from `src/CMakeLists.txt` by f515fbb356597ac56324c756c96578e00f0037f2:

```
-# Configuration file for pkg-config
-configure_file(
-  ${CMAKE_SOURCE_DIR}/scripts/argos3.pc.in
-  ${CMAKE_BINARY_DIR}/argos3_${ARGOS_BUILD_FOR}.pc
-  @ONLY)
-install(FILES ${CMAKE_BINARY_DIR}/argos3_${ARGOS_BUILD_FOR}.pc DESTINATION lib/pkgconfig)
```
which seems to be the source of the issue. I think? There are present in c0ff091e9f037b022281cec6080c78fa3bd57879 and not in later commits after FindARGoS was updated.

Was this an accidental deletion, or was this the old mechanism supposed to be replaced by something new ?

closed 2021-06-14 by ilpincy

jharwell, 2021-06-14:
In that case, the examples should be updated--they still say that if FindARGoS fails to check pkg-config settings.

ilpincy, 2021-06-14:
I am not sure I understand the issue. Can you show me what's going wrong on your side? The correct code is what is reported in the examples [here](https://github.com/ilpincy/argos3-examples/blob/master/CMakeLists.txt#L22):

```
# Find the ARGoS package, make sure to save the ARGoS prefix
find_package(ARGoS REQUIRED)
include_directories(${CMAKE_SOURCE_DIR} ${ARGOS_INCLUDE_DIRS})
link_directories(${ARGOS_LIBRARY_DIR})
link_libraries(${ARGOS_LDFLAGS})
```

jharwell, 2021-06-14:
The code is correct, but the README https://github.com/ilpincy/argos3-examples/blob/master/README still contains: 

When you launch one of the two 'cmake' commands, you might get an error like 'package argos3_simulator' not found. This error has two possible causes:
1. You haven't installed ARGoS3 system-wide. To fix this, install   ARGoS3 through the packages or with the command 'sudo make install'   in the ARGoS3 build directory.
2. pkg-config can't find the file argos3_simulator.pc. When you   install ARGoS3 from a package, this file is located in   /usr/lib/pkgconfig/. When you install ARGoS3 from the sources, this   file is located in $CMAKE_INSTALL_PREFIX/lib/pkgconfig/. If you   don't set $CMAKE_INSTALL_PREFIX explicitly before compiling, by   default it is set to /usr/local.  If the file is present and   pkg-config can't find it, add the directory where it's located to   the environment variable PKG_CONFIG_PATH, e.g.

$ export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig

which is where my confusion came from--I checked the README first.

ilpincy, 2021-06-14:
Well spotted - I forgot to update the README. Thanks, I'll change it!