ilpincy/argos3 #178: Compiling the e-puck plugin  for argos3 59

teksander, 2021-06-25:
Hi Carlo,
In the argos e-puck plugin I changed some lines in the file https://github.com/demiurge-project/argos3-epuck/blob/master/src/cmake/ARGoSBuildChecks.cmake, according to what is required in argos59.

Instead of pkg_check_modules, I changed to:

  set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${ARGOS_PREFIX}/share/argos3/cmake)
  include_directories(${CMAKE_SOURCE_DIR} ${ARGOS_INCLUDE_DIRS})
  link_directories(${ARGOS_LIBRARY_DIR})
  link_libraries(${ARGOS_LDFLAGS})

and also replaced this include:

if(ARGOS_BUILD_FOR_SIMULATOR)
include(FindARGoSQTOpenGL)
endif(ARGOS_BUILD_FOR_SIMULATOR)

The plugin works (actuators and sensors), however the E-pucks are not rendered in the GUI. Do you know where else changes might be required?
Cheers

ilpincy, 2021-06-25:
I'm not sure, but I'll have a look!

allsey87, 2021-06-28:
The problem is probably related to the variable `ARGOS_COMPILE_QTOPENGL`, which is no longer used but is still declared in [config.h](https://github.com/ilpincy/argos3/blob/f2b2d0d7b5dac6500ae34bb69d03f18ae4502257/src/core/config.h.in#L39).

I haven't checked this, but I think the solution will be to update [SimLibrary.cmake](https://github.com/demiurge-project/argos3-epuck/blob/master/src/plugins/robots/e-puck/SimLibrary.cmake#L24) to check against the variable `ARGOS_QTOPENGL_FOUND` instead of `ARGOS_COMPILE_QTOPENGL`.

teksander, 2021-06-30:
Thank you @allsey87 !

The two changes I mentioned to [ARGoSBuildChecks.cmake](https://github.com/demiurge-project/argos3-epuck/blob/master/src/cmake/ARGoSBuildChecks.cmake) and the 6 changes to [SimLibrary.cmake](https://github.com/demiurge-project/argos3-epuck/blob/master/src/plugins/robots/e-puck/SimLibrary.cmake#L24) fixed the issue. So far, everything on the e-puck plugin seems to be working in argos59

closed 2021-06-30 by teksander

ilpincy, 2021-06-30:
Thanks @allsey87! 💯