ilpincy/argos3 #179: No package 'argos3_simulator' found

kashif-hub, 2021-07-08:
cmake ../src
-- Checking for module 'argos3_simulator'
--   No package 'argos3_simulator' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/ARGoSBuildChecks.cmake:6 (pkg_check_modules)
  CMakeLists.txt:31 (include)

-- Configuring incomplete, errors occurred!
See also "/home/ffmpeg/software/argos3-epuck/build/CMakeFiles/CMakeOutput.log".
[image]

allsey87, 2021-07-08:
Could you provide a link to the version of the e-puck plugin that you are trying to compile?

ilpincy, 2021-07-08:
This should not be here, it's an issue with the e-puck plugin. I'll mark it as invalid and close it.

closed 2021-07-08 by ilpincy