ilpincy/argos3 #198: Increasing the # cameras in ARGoS

jharwell, 2022-01-24:
Hello,

I'm wanting to add this because I want to generate some really nice videos showing robots at 3 different levels of detail (far away as dots, medium distance, and then close up in detail) as I interpolate around the arena. Say I want 36 cameras: 36 = interpolating between 3 sets of 12 cameras equally spaced around the unit circle (each set is at a different Z and the circle all 12 cameras in the set are on is different). 

Would there be any issue in changing the # of cameras from 12 to say 36 here:

https://github.com/ilpincy/argos3/blob/b19c7539c427b9232d0f87a567ad293b67167dbb/src/plugins/simulator/visualizations/qt-opengl/qtopengl_camera.h#L137

The GUI still shows the original 12 cameras, but ARGoS now allows you to manipulate all 36 cameras in XML. There doesn't appear to be any other ill effects, but I wanted to be sure before I opened a pull request. 

A "better" solution I think would be to make this an XML parameter, and to use `std::vector` instead of std::array, but I don't know if that's too invasive of a change.

Would you be open to merging either of these if I implement and open a pull request?

allsey87, 2022-01-24:
Hi @jharwell, while what you are proposing here seems reasonable, I am not sure whether it is useful to enough people to warrant the development time. Personally, I would lean towards categorizing this as changes that can just be left in a local branch. I often do this myself too when I need to tweak the output of ARGoS in a certain way.

ilpincy, 2022-01-24:
It's a neat feature, if we had to implement it "the right way" the GUI would start with a single camera and + and - buttons would allow the user to manipulate the number of cameras. 

Implementing it halfway, i.e., allowing for more cameras in the XML but only visualizing 12, will lead to bugs and inconsistency.

jharwell, 2022-01-24:
Could I just make all cameras (however many there are) viewable in the GUI? i.e., have 36 camera buttons to switch between cameras instead of 12?

ilpincy, 2022-01-24:
No, that would clutter the GUI too much. There's no general use for so many cameras.

jharwell, 2022-01-24:
That's fair. I'll keep this in my fork then.

closed 2022-01-24 by jharwell