ilpincy/argos3 #210: Black screen

abhishekranaji, 2022-05-11:
On running Argos it is showing black screen as below

Device Raspberry pi 4
Os debian 11 based linux

[image]

dcat52, 2022-05-11:
Hello @abhishekranaji, I have seen this before but I don't quite remember the exact solution. Are there some printouts regarding QT in the terminal?
From my memory, it has to do with the QT wireframe box sides not being transparent. A quick test of this can be done by zooming into the black box and seeing if you can make it inside of it.

abhishekranaji, 2022-05-11:
~/argos3-examples $ argos3 -c experiments/diffusion_10.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d_magnetism.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_builderbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_kilobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d_floor.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_block.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_prototype.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d_gravity.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_drone.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d_srocs.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pipuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_diffusion/libfootbot_diffusion.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)

I have tried zooming in but cant see any thing, Here is the dump for diffuesion_10.argos
@dcat52

pefribeiro, 2025-10-15:
Same problem here running this with with VNC under Ubuntu 22.04, so software rendering. Screenshot below with `experiments/diffusion_1.argos`.

[image]

glxinfo -B output:

```
name of display: :1
display: :1  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa (0xffffffff)
    Device: llvmpipe (LLVM 15.0.7, 128 bits) (0xffffffff)
    Version: 23.2.1
    Accelerated: no
    Video memory: 32045MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 31 MB, largest block: 31 MB
    VBO free aux. memory - total: 28798 MB, largest block: 28798 MB
    Texture free memory - total: 31 MB, largest block: 31 MB
    Texture free aux. memory - total: 28798 MB, largest block: 28798 MB
    Renderbuffer free memory - total: 31 MB, largest block: 31 MB
    Renderbuffer free aux. memory - total: 28798 MB, largest block: 28798 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 4294967037 MB
    Total available memory: 31786 MB
    Currently available dedicated video memory: 31 MB
OpenGL vendor string: Mesa
OpenGL renderer string: llvmpipe (LLVM 15.0.7, 128 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.5 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL shading language version string: 4.50
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

Does it require a specific version of OpenGL? I've also tried setting LIBGL_ALWAYS_SOFTWARE=1

Output from argos3 below:
```
$ argos3 -c experiments/diffusion_1.argos 
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_drone.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_prototype.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d_srocs.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_builderbot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d_floor.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d_magnetism.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_block.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_pipuck.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d_gravity.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_diffusion/libfootbot_diffusion.so"
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ubuntu'
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
```