ilpincy/argos3 #213: Install MacOS glut not found

ffelten, 2022-09-21:
Hello,

I am trying to install Argos on my mac. I have installed all required packages using brew and everything seems good. However, when I try to run `make`, I get errors like this:
`ld: library not found for -lglut`

I managed to track the problem down to `build/.../CMakeFiles/...dir/link.txt` files (in various places).

By manually replacing `-lglut` with `-framework GLUT`, I managed to compile. Yet I think this problem might reoccur since these look like autogenerated files. The problem is that I have no clue on where to look to fix it at the root.

Cheers,

ilpincy, 2022-09-21:
Looks like an issue I'm my way to detect OpenGL. Can you give me some details about your system? Mostly OS version and architecture.

ffelten, 2022-09-21:
MacOS Monterey v12.3.1, Intel chip.
XCode 13.3 and clang-15.

ilpincy, 2022-09-21:
Can you show me the output of CMake when you run
```
$ rm -rf build
$ mkdir build
$ cd build
$ cmake ../src
```

ffelten, 2022-09-21:
```
-- The C compiler identification is Clang 15.0.0
-- The CXX compiler identification is Clang 15.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/opt/llvm/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/opt/llvm/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found DLFCN: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/lib/libdl.tbd
-- Found Pthreads: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/lib/libpthread.tbd
-- Found FreeImage:
-- Could NOT find GooglePerfTools (missing: GOOGLEPERFTOOLS_LIBRARY GOOGLEPERFTOOLS_INCLUDE_DIR)
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.9.5") found components: doxygen dot
-- Found ASCIIDoc: /usr/local/bin/asciidoc
-- Found OpenGL: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework
-- Found GLUT: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found Qt6: version 6.3.1
-- Found Qt6Widgets: version 6.3.1
-- Found Qt6Gui: version 6.3.1
-- Found Qt6OpenGLWidgets: version 6.3.1
-- Found Lua: /usr/local/lib/liblua5.4.dylib (found version "5.4.4")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/florian.felten/Documents/argos3/build
```

ilpincy, 2022-09-21:
The message
```
-- Found GLUT: TRUE
```
is different from mine. On my system I get the full path to the framework rather than `TRUE`. What's your CMake version?

ilpincy, 2022-09-21:
I dug a bit deeper – on my system everything works as usual, so it's difficult for me to pinpoint the issue on your end. Do you have multiple versions of GLUT installed, by chance? Maybe `freeglut` confuses CMake?

ffelten, 2022-09-21:
CMake version 3.24.2

Output for GLUT:
```
pkgcfg_lib_GLUT_glut:FILEPATH=/usr/local/Cellar/freeglut/3.2.2/lib/libglut.dylib
//Details about finding GLUT
FIND_PACKAGE_MESSAGE_DETAILS_GLUT:INTERNAL=[1][v()]
GLUT_CFLAGS:INTERNAL=-DFREEGLUT_STATIC;-I/usr/local/Cellar/freeglut/3.2.2/include
GLUT_CFLAGS_I:INTERNAL=
GLUT_CFLAGS_OTHER:INTERNAL=-DFREEGLUT_STATIC
GLUT_FOUND:INTERNAL=1
GLUT_INCLUDEDIR:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/include
GLUT_INCLUDE_DIRS:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/include
GLUT_LDFLAGS:INTERNAL=-L/usr/local/Cellar/freeglut/3.2.2/lib;-lglut
GLUT_LDFLAGS_OTHER:INTERNAL=
GLUT_LIBDIR:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/lib
GLUT_LIBRARIES:INTERNAL=glut
GLUT_LIBRARY_DIRS:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/lib
GLUT_LIBS:INTERNAL=
GLUT_LIBS_L:INTERNAL=
GLUT_LIBS_OTHER:INTERNAL=
GLUT_LIBS_PATHS:INTERNAL=
GLUT_MODULE_NAME:INTERNAL=glut
GLUT_PREFIX:INTERNAL=/usr/local/Cellar/freeglut/3.2.2
GLUT_STATIC_CFLAGS:INTERNAL=-DFREEGLUT_STATIC;-I/usr/local/Cellar/freeglut/3.2.2/include
GLUT_STATIC_CFLAGS_I:INTERNAL=
GLUT_STATIC_CFLAGS_OTHER:INTERNAL=-DFREEGLUT_STATIC
GLUT_STATIC_INCLUDE_DIRS:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/include
GLUT_STATIC_LDFLAGS:INTERNAL=-L/usr/local/Cellar/freeglut/3.2.2/lib;-lglut;-lX11;-lXxf86vm;-lXrandr;-lGL;-lm
GLUT_STATIC_LDFLAGS_OTHER:INTERNAL=
GLUT_STATIC_LIBDIR:INTERNAL=
GLUT_STATIC_LIBRARIES:INTERNAL=glut;X11;Xxf86vm;Xrandr;GL;m
GLUT_STATIC_LIBRARY_DIRS:INTERNAL=/usr/local/Cellar/freeglut/3.2.2/lib
GLUT_STATIC_LIBS:INTERNAL=
GLUT_STATIC_LIBS_L:INTERNAL=
GLUT_STATIC_LIBS_OTHER:INTERNAL=
GLUT_STATIC_LIBS_PATHS:INTERNAL=
GLUT_VERSION:INTERNAL=3.2.2
GLUT_glut_INCLUDEDIR:INTERNAL=
GLUT_glut_LIBDIR:INTERNAL=
GLUT_glut_PREFIX:INTERNAL=
GLUT_glut_VERSION:INTERNAL=
__pkg_config_arguments_GLUT:INTERNAL=QUIET;glut
__pkg_config_checked_GLUT:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_GLUT_glut
pkgcfg_lib_GLUT_glut-ADVANCED:INTERNAL=1
```

ilpincy, 2022-09-21:
Yes, you have freeglut installed and that is confusing CMake. You can fix that in a couple of ways: you can temporarily uninstall freeglut, or you can manually set the correct path for GLUT_LIBRARIES in CMake.

closed 2022-09-21 by ilpincy

ffelten, 2022-09-21:
Awesome! Thank you!