ilpincy/argos3 #216: Bugs and usability of ARGoS drone

jharwell, 2023-01-03:
If I create a drone at (3,3,0) for example, and call `SetTargetPosition()` with `(5,4,0)`, and then wait, the drone does not go to that location. Instead, it goes to ~`(3,10,0)`. The code doesn't have any documentation, but I _think_ `SetTargetPosition()` is relative to where the drone was created at, from reading the Lua docs. But that's not right either, because then the drone should end up at `(8,7,0)`, which it doesn't. What's the correct behavior here, and does the code implement it?

While this is my immediate issue with using the drone, there are others:
- Other robots have a constructor which takes the ID and the controller ID as the first two arguments, and the drone does not, so you can't change what it is looking for the controller as, as far as I can tell. This also makes it difficult to create new robots appropriately in the loop functions.
- You can't construct a drone at an arbitrary location in the arena, which you can do with other robots (I use this ability a lot for other robots).
- There are no C++ (or even Lua) examples to look at in argos3-examples. There is the README in the ARGoS source, but that only has Lua snippets.

I was able to patch the drone and add a new constructor which allows me to create drones. I would be happy to create the lacking examples, but I can't figure out how to use it to the point where I could create some :-(.

Overall, my experience thus far trying to use the drone is much different than the other robots, which worked as I expected out of the box. It seems much less mature, and marking it as "usable" like the foot-bot doesn't seem right, because of the issues above. Maybe "beta" would be better?

allsey87, 2023-01-04:
To the best of my knowledge, the drone works well and is currently being heavily used for research at IRIDIA. It is also one of the robots in ARGoS which allows you to move code with almost no changes between simulation and [its hardware implementation](https://github.com/iridia-ulb/argos3-drone). 

The issue you raise with `SetTargetPosition` may be related to the initial orientation of the drone, e.g., if it is rotated around its z-axis, and you supply `SetTargetPosition` with a vector such as (1,0,0), it will move along both the x-axis and y-axis with respect to its orientation. This is all consistent with the hardware which is based on MAVLink/PX4 flight computer.

Regarding the constructors for the drone, there isn't really a standard for this. Most robots do things a little bit differently and support different sets of features via different constructors. I opted to use a single XML-based `Init` method since keeping two initialization approaches consistent was becoming a headache to maintain. You can, however, use the drone from a loop function by building a custom `TConfigurationNode` in your code and passing it to `CDroneEntity::Init`. I agree though that there could be more examples on how to do this.

Whether a robot is marked as `beta` or `usable` doesn't really mean much at the end of the day. It can be changed, but to be honest, I don't agree that the drone is in any less of a usable state than the other robots.

jharwell, 2023-01-04:
re: `SetTargetPosition()` -- that helps!

I understand that from a technical perspective the drone is 100% mature and usable. However, from a broader perspective, such as that of a technically skilled ARGoS user, it is much less usable, because of the lack of documentation. If I'm being honest, if I didn't already have so much ARGoS tooling set up for 2D robots, and I was coming in cold to try to get the drone to work, I would have given up in short order without any docs, examples, or code comments. I would imagine other users unfamiliar with the technical details of the drone would have similar experiences. 

Again, I would be happy to add docs, examples, and code comments and open a pull request, but I need _some_ kind of starting point--is there a self-contained C++ something from IRIDIA using the drone that I could look at?

allsey87, 2023-01-04:
There is a technical report for the drone here: https://iridia.ulb.ac.be/IridiaTrSeries/link/IridiaTr2022-002.pdf, if you need more information, I would reach out to either @freedomcondor or @sinanOguz