ilpincy/argos3 #217: footbot_encoder can not be reading.

yunxiaoshan, 2023-01-27:
Hi,
I am trying to add a footbot_encoder sensor on a footbot robot to get the wheel speed. What I do like this:
` const CCI_DifferentialSteeringSensor::SReading& tEncoderReads = m_pcEncoder->GetReading();`
`cout << tEncoderReads.VelocityRightWheel <<endl;`
The build process can work. However, When I launch the simulator, the segmentation fault will happen. I figure out that the error is the `cout.` When the `cout` is commented, everything goes well. 
The first question is am I doing that in the right way? May I have a solution to print the encoder readings?
Thank you!
-shan

jharwell, 2023-01-27:
I think this is better asked in the forums: http://www.argos-sim.info/index.php. Without seeing more of your code, I can only guess that `m_pcEncoder` is NULL. Did you (a) put the differential steering sensor in the XML, and (b) call `GetSensor()` as shown in the examples (https://www.argos-sim.info/examples.php)?

yunxiaoshan, 2023-01-28:
really appreciate your fast response. I found my problem, as you advice of (b), I forget to get sensor. For the forums, I do not know why it cannot be browsed and shows SQL ERROR [ mysqli ] error. anyway, thank you and I will try to figure out the error.

closed 2023-01-28 by yunxiaoshan

ilpincy, 2023-01-29:
I didn't notice the db error in the forum, thanks for pointing that out! The forum is back in action.